KiCad/kicad-footprints #2473: SMD Feedthrough Capacitors

jamesrlucas, 2020-09-18:
I am looking to use some feedthrough capacitors in some design I am working on, however there are currently no footprints in the libraries. Before I start creating & submitting footprints, I'd like to ask for opinions on how to go about it.

Example devices

[Murata 0603](http://www.farnell.com/datasheets/90767.pdf)
[TDK 0603 & 0805](http://www.farnell.com/datasheets/2942825.pdf)
[Murata 0805](http://www.farnell.com/datasheets/608981.pdf)
[AVX 0805 & 1206](http://www.farnell.com/datasheets/2097719.pdf)
[Knowles 0805, 1206, 1806 & 1812](https://www.knowlescapacitors.com/getattachment/982649d9-f72f-4bee-a557-e7285e5005bc/E01-300mA.aspx)
[Murata 1206](http://www.farnell.com/datasheets/1902092.pdf)

All have slightly different recommendations on land patterns (if any!)

Questions:

1: Does anyone have access to the IPC recommendations? (if any!) - I have no idea if feedthrough caps are included or not.
2: Are these footprints we could script?
3. Would a single generic footprint be OK, or should we use manufacturer-specific footprints?
4. What library to put them in - Capacitor_SMD or Filter?

Thanks,

James

cpresser, 2020-09-19:
> Questions:
> 
> 1: Does anyone have access to the IPC recommendations? (if any!) - I have no idea if feedthrough caps are included or not.

See https://github.com/pointhi/kicad-footprint-generator/pull/438, at least one outdated `IPC-SM-782` document is linked there.
That generator is also partially related, it does RLC footprints.

> 2: Are these footprints we could script?

Yes but .... AFAIK there is no script that can handle it. It needs to be created. Those parts a rather special, so it is not mandatory to script them.

> 3. Would a single generic footprint be OK, or should we use manufacturer-specific footprints?

Depends. I would start comparing them and see if there can be a consensus footprint.

> 4. What library to put them in - Capacitor_SMD or Filter?

Everything is a filter if you are brave enough!
`Capacitor_SMD` sounds better to me, but I have no arguments.

cpresser, 2020-09-19:
From the Murata 1206 Datasheet. Quite interesting, since there is more paste than copper area:
[image]
[image]