KiCad/kicad-footprints #2476: DO-27 Pad Hole Size Too Small

fatalelieberi, 2020-09-20:
Hi, I just want to report that the standard hole size for DO-27 diodes is too small. It should be, at minimum 1.3mm, or ideally 1.4mm, instead of 1.2mm in the standard Kicad library. Many manufacturers specify the lead diameter to be 1.3mm at maximum.

Manufacturer data sheet
[image]

Kicad Library
[image]

chschlue, 2020-09-20:
JEDEC DO-201AA (formerly DO-27) has 0.91mm max lead diameter.
What you're looking at is probably in fact a DO-201AD package (1.42mm max.)

closed 2020-09-20 by chschlue

fatalelieberi, 2020-09-20:
Well, that's weird. It seems that some manufacturers use DO-27 package name for some types even though they have dimensions as specified by DO-201AD, such as MBR5200, SK306, SR520-SR5100 diodes.

chschlue, 2020-09-20:
Yes, we are frequently faced with naming issues like that.
But simply increasing the actual DO-27 FP to 1.5mm hole diameter (max lead + 0.2mm) isn't a sensible solution.