KiCad/kicad-footprints #2484: _Handsolder footprints in v6

cpresser, 2020-09-22:
In https://github.com/KiCad/kicad-footprints/pull/2280#issuecomment-696971187 the question came up how to handle the `_Handsolder` footprints.

The problem is that they are sprinkled all over the lib. And they don't have clear rules on how to build them. Thus, they vastly differ.

I think we should keep them. KiCad is used a lot by hobbyists and beginners - most of those people are afraid of any SMD/SMT stuff. With dedicated footprints they have an easier start into the SMT world.
But I think we should devise some kind of ruleset for them. Perhaps increased courtyard, and a explicit rule for pad elongation.

chschlue, 2020-09-22:
Picking up your 0805 example:
If I'm not mistaken, a level A variant has an even longer pad than the current 1.15mm HandSolder one.
Why not go with the rule set we already have? (And perhaps add HandSolder to the tags or something.)

evanshultz, 2020-09-22:
That sounds like a great proposal to me. Level A doesn't say it's used for hand solder, but it does mention wave soldering and it might make sense to put that into the keywords as well.
[image]

calebreister, 2020-09-23:
Personally, I've never had trouble hand-soldering the regular footprints down to 0603. I did handsolder/rework 0402 once, but I wouldn't wish that upon anyone.