KiCad/kicad-footprints #2486: Vishay D0-220AA-SMP diode

Almisuifre, 2020-09-23:
Good evening,

Would it be possible to add the Vishay D0-220AA-SMP diode to the library.

Attached is a zip with the datasheet, the 3D diode (modeled by me under FreeCAD) and the footprint?
[Vishay-SS1P3L-SS1P4L.zip](https://github.com/KiCad/kicad-footprints/files/5271751/Vishay-SS1P3L-SS1P4L.zip)

You have permission to modify so that it meets your standards.

Thank you

evanshultz, 2020-09-25:
@Almisuifre 
Thanks for making these library assets and submitting them.

We rely on the community to grow and fix the library, with all libraries being volunteers, so we need a bit more help from to be able to use these. Please create a PR for the footprint in this repo and the 3D model at https://github.com/KiCad/kicad-packages3D with the 3D model source files at https://github.com/KiCad/kicad-packages3D-source. We are moving to GitLab soon so if there isn't time to review and merge these before moving, which seems likely, there will be similar repos on GitLab soon where you can submit the files again.

Almisuifre, 2020-09-25:
Thank you for your reply. I will take care of making both requests in parallel. I also insert the link to here because it is the starting point of this sharing.
In the future, if I understand correctly:

> kicad-footprint: Pull request -> message + pretty
> kicad-package3D: Pull request -> message + 3dshapes (step)
> kicad-package3D-sources: Pull request -> message + 3dshapes (FreeCAD)

If this is correct I will take it into account for other components.
Ok for the migration to GitLAB. It was a bit predictable ...